MariwanJ/Design456 #3: Related links

luzpaz, 2021-01-31:
### List of related informative links 
* [Draft: subelements move, working directly on Shape](https://github.com/FreeCAD/FreeCAD/pull/3716) by @carlopav
* [Ticket 3353: Direct modeling tools](https://tracker.freecadweb.org/view.php?id=3353)
* GSoC 2017 proposal [Direct Modeling tools](https://wiki.freecadweb.org/Direct_modeling_tools)

MariwanJ, 2021-02-02:
If you look at 123D Design, you can do funny things with Tweak. So, even if the tools is good, but you have to be careful. 
That is the same for this proposal. Even if we have it .. user must be careful how to use it and the limitation. 
Still I didn't have time to look at the code.

MariwanJ, 2021-02-02:
Unfortunately,  Due to my lack of experience, I cannot get the code working.
I cannot make it as a class. I tried many things .. fails. 
To not lose this time on only one function that I don't understand and it is fragile .. I will wait this this and put it in another sub branch.
If you can make it as a class that I call I will appreciate it

luzpaz, 2021-02-02:
@MariwanJ wrong ticket :stuck_out_tongue_closed_eyes:

carlopav, 2021-02-02:
If you want in available for a jitsy meet to help you get quickly more confidence with the code ;)

MariwanJ, 2021-02-02:
Dear Carlopav. 
Thank you very much. The problem is that I am new to python. It is not the difficulty of the code.  I wish to spend some more time learning the FreeCAD python. If that didn't fix my problem I will appreciate your help. But to not let you waste your time without a big gain, I wish we wait. THANK YOU very much that you care.
I will continue adding small small instructions and shape creation to the workbench.. during this journey I will learn more

closed 2021-04-25 by MariwanJ