MariwanJ/Design456 #5: Code styling and readability

luzpaz, 2021-02-11:
Some recommendations for code styling:  
https://wiki.freecadweb.org/Python

MariwanJ, 2021-02-12:
mm. 
I really didn't put so much effort on documenting, cleaning code ..etc. I wish just to make things working. 
I have a big problem with Notepad++ .. Indent problem always. Some suggestion Visual Studio .. I will try to use asap.
I will read that and try to understand what I am doing wrong. If you have example from my code would make my life easier. Thank you very much

MariwanJ, 2021-02-13:
I hope that it is fixed now since I use Visual Studio Code and many plugin that help in having it correct .. Please let me know if it is still bad style. I close this for now.

closed 2021-02-13 by MariwanJ