MariwanJ/Design456 #7: Tweak

MariwanJ, 2021-02-14:
@carlopav 
1-The face it takes to modify is not what you click. This should be fixed. 
2-Trying to re-use the tool on the same face it affect the old object. It shouldn't be like that. 
3- I think the new object should be merged and fined to make them as one part. 
Look at the picture.

carlopav, 2021-02-14:
The tool works like this only if the given object is a Part::Feature. If this is not the case, the object is converted to a Part::Feature and the new shape is applied. If you change that you will have a tool that breaks objects, without giving the user the possibility to go back... i'm not sure this is a good design option.
Anyway you are free to change it as you prefer. ATM I'm still out of time to dedicate to the project...

MariwanJ, 2021-02-14:
ok thanks. I will try to make it better when I have time.

closed 2021-02-14 by MariwanJ