MariwanJ/Design456 #10: Is this workbench planning to tackle tickets 3353 and 1367 ?

luzpaz, 2021-03-02:
See https://wiki.freecadweb.org/Direct_modeling_tools#Additional_Information  
https://tracker.freecadweb.org/view.php?id=3353  
https://tracker.freecadweb.org/view.php?id=1367

MariwanJ, 2021-03-02:
Thanks for your question. 
Short answer : Yes, that is my goal in the Design456 WB. 
If you have been using 123D Design you feel the freedom of reforming shapes. Creating shapes quickly with dimensions.
Tweak, smart scale, extrude and all other tools are easy to use, understand. 
Most of the tools in Draft or Part ..etc are difficult to use, or understand, IMHO.
What I didn't understand yet in freecad is the structure of the objects. They seems to be READ only. Whenever you want to change anything .. you need to make a copy of the shape to reproduce the object. (That is how I understood to the time of writing this comment). 
My aim is to make all the necessary changes to the tools Part, and Draft have. 
Take an example: Why do Extrude cannot continue extrude an old object? I cannot understand that.
My question to you: I am not familiar with "Tracker" pages. Would you let me know what is the purpose of that page and if I should be there? 
I wish and will continue to develop Design456 to be THE tools for "Direct modeling tools". 
I have experience in cpp that shouldn't be a problem .. if that requires. But I am sticking now on the python as I learn it and might find the solution using it. I don't know If I must think in cpp. 
Please let me know what you think?  
Note: 
I know that I need a better project planning. But at the moment I wish to make some tools that might be easy and I learn also in the way of making them. After learning the internal API , I might make my steps more structured and more targeted to specific solutions (for ex  2D drawing first, or 3D drawing first ..etc)
Please share with me what you are thinking and wish I do?

MariwanJ, 2021-06-21:
Some progress. I have now the smart scale somehow working. I need to fix some issues but It works. 
[https://youtu.be/GmWSs9Q36YA
](url)

closed 2021-07-16 by MariwanJ