MariwanJ/Design456 #11: Problem on first try to open Workbench - polyhedrons missing

wagnerlip, 2021-03-06:
21:54:26  No module named 'polyhedrons'
21:54:26  Traceback (most recent call last):
  File "<string>", line 55, in Initialize
  File "C:\FreeCad19A\Mod\Design456\Design456_Part.py", line 48, in <module>
    class Design456_Part:
  File "C:\FreeCad19A\Mod\Design456\Design456_Part.py", line 49, in Design456_Part
    import polyhedrons

21:55:39  module 'FreeCAD' has no attribute 'DraftWorkingPlane'Error: Draft activation failed, Draft will not work as expected.

MariwanJ, 2021-03-06:
Please you cannot just download the zip .. It wouldn't work as it has dependecy on other code. 
use the following command
git clone --recurse-submodules -j8 https://github.com/MariwanJ/Design456

luzpaz, 2021-03-06:
cc @joelgraff  
Hi Joel, quick question: how does the addon manager deal with the trails submodule dependencies (in your case the pivy-tracker and the python tools). What's the process? Do you update them and then make a commit and that's what addon manager sees?

wagnerlip, 2021-03-06:
I never heard of "git" command on Windows.  I will try.

MariwanJ, 2021-03-06:
You have to install it. Please read the README. There is some more info. 
Download git from 

https://git-scm.com/download/win

MariwanJ, 2021-03-06:
@Luzpaz
Isn't better to remove the submodule and added permanently to my repo? This is the first time someone fail to download the WB but in the future might make it a big issue. 
I think I need to combined and remove it as a sub-module? 
Please let me know what you think?

luzpaz, 2021-03-06:
waiting on response... will get back to you on this

wagnerlip, 2021-03-06:
So, everybody trying to install this WorkBench in FreeCad will need to download and install GIT ?
Some installations of FreeCad are in companies PCs, where the local server has severe restrictions and firewalls against strange things.  FreeCad by itself, not requiring installation at all (just unzip and run), is a candy on those installations.  Anything else will be a burdensome.  I simply can not download and install GIT for that reason.  But I can in fact download a ZIP file and unpack in a certain \MOD folder.   I can also think about the ones with a PC running windows standalone, without internet connection (for several reasons), using a pen-drive to more the ZIP file is a neat solution, GIT will not work at all in those situations.   As a matter of fact I know several PCs Win10 running FreeCad and other softwares serving 3D printers, CNC or Lasers Cutters, not even connected to the company local area network.

Things to remember:
1) Not everybody run Linux, it is less than 2% of Desktop Op System market-share
2) Not everybody is connected to the Internet
3) There are several firewalls and restrictions on companies PCs
4) The term "recurse submodules" may not be understandable for millions of users
5) Not everybody is a programmer, and not all CAD users will understand programming issues
6) I write code since DOS 3.1 (198ish) and always keep in mind: what I know and think is not popular knowledge, don't think the regular user understand what you think and what you ask him to do
7) The easier you make your installation packages, more people will use it

joelgraff, 2021-03-07:
Honestly, I don't really know.  I never got clear direction on that when I
posted about it in the forums - or if I did, I missed it.

From what I understand, submodules *should* be supported.  All that's
required is to add a single parameter to the 'git clone' command, and I was
under the impression that it happened.  But now, I really don't know.  I
will try it when I can, but I think Trails is still broken in the Addon
Manager.  Maybe Addon Manager is pulling from the wrong branch?  That was
another issue that I don't know if it was ever sorted out.

I've debated just adding the submodules back to the Trails repo directly.
Maybe that should just happen.  I have other projects that depend on those
modules, but since I'm the only one working on them, kinda hard to claim
it's imperative they remain as submodules.

Anyway, I'm reluctant to eliminate the submodules as it appears to me
there's no reason why it ought not to work.  So until there's some clear
direction from above on that, I really don't know what to say.

*Joel Graff, P.E.*

*LinkedIn:* https://www.linkedin.com/in/joelcgraff

MariwanJ, 2021-03-07:
OK. thanks. 
I will remove that as sub-module. Design456 will be responsible for further development of that sub-module. 
I really didn't know what to choose. I got that answer (to be as sub-module) from the forum. But I see now it is a problem, at least for now. 
I wouldn't recommend it as sub-module either as it make it more complicated. I cannot either use the sub-modules as it is .. I have to change many thing in them. 
So, if I cannot bring it like I did for Draft, I will merged to my Design456.
I hope I fix that today.

luzpaz, 2021-03-07:
wait..still getting a reply on the submodule thing.

joelgraff, 2021-03-07:
I'm really not inclined to roll pivy-trackers into the Trails workbench.
It's a complex enough project on its own and it's meant to stand alone as
an engineering tool design framework.  It becomes difficult for me to
maintain if I roll it into Trails as an integral part of the codebase.

Handling submodules should be a well-solved problem that shouldn't cause
anyone grief.  I don't know why the Addon Manager doesn't handle it
properly, but I'm open to doing what it takes so submodules are properly
supported.

*Joel Graff, P.E.*

*LinkedIn:* https://www.linkedin.com/in/joelcgraff

MariwanJ, 2021-03-08:
I will definitely remove this submodule and put the file in the main repo. I changed alot and there is no meaning of having other files from his WB which I don't use. I will keep the header, icons ..etc but they will be merged for ever with the WB.
And I will be avoiding adding others repo as there are many people that cannot treat sub modules.