MariwanJ/Design456 #12: Subtract 2 2-d Faces or Combine 2 2-d Faces shift X and Y

wagnerlip, 2021-03-06:
Subtracting or Combining 2 2-D faces move the result to a grid position that is the addition of X and Y from the two faces.
Example:  A square at X20, Y10, Z5 and a Circle at X22, Y5, Z5, when combining will result the combined object to move to X42, Y15, Z5. 

Also, if after the subtracting or combining, you press Ctrl-Z  to cancel the operation, it cancel, but one of the faces becomes extruded.

MariwanJ, 2021-03-06:
I wrote the problems I have in some of the commands. Yes, UNDO-REDO doesn't work at all in my WB. 
I have not time to fix that. 
My ROADMAP  is to make the commands available and then fix these kind of things. You need to be careful when you use the WB. 
One person working on a big project will not be enough to do all the tasks unfortunately .. I am aware of that. 
Regarding the dimension: I know that also. I haven't time to decide which position I should use when I subtract or merge  two objects. 
Let say object one has 1,1,0 and object 2 has 30,30,0  .. Which one you will choose ? The subtracted object or the subtractor? And when the object you deleted a part from will no longer has the same dimension and Base point. What should you choose? 
It is kinda asthmatically thinking that I couldn't put the required time to figure out what is the best. 
Please feel free to suggest some. 
Think about what you should do when you subtract  shape1 - shape2 --> where none of the Base will be suitable for the new object. 
Thanks for your report.

MariwanJ, 2021-03-06:
It is extruded objects where are used to make add, subtract and common. So that is why you see it. Look at the code please to understand.

wagnerlip, 2021-03-06:
> Let say object one has 1,1,0 and object 2 has 30,30,0 .. Which one you will choose ? The subtracted object or the subtractor? And when the object you deleted a part from will no longer has the same [...]

If object A is at 1,1,0 and B is at 30,30,0, they possible do not overlap, and not overlapping how you cut/union?
I always thought that a circle B cutting circle A, will cut exactly where you see the overlap, and the resulting object A will still in the same place, just missing a bite.  That is, object B will disappear taking along its shape from A.  There is no need to move the pizza to the other table just because you took a slice, pizza still in the same place, inside its box, missing a slice.

wagnerlip, 2021-03-06:
> It is extruded objects where are used to make add, subtract and common. So that is why you see it. Look at the code please to understand.

I thought you could do the operation within the sketch 2D objects, no needing to extruded them to 3D first.  Either way, the 3D conversion should be used as temporary and only during the operation, not leaving any 3D behind because of a Ctrl-Z.

wagnerlip, 2021-03-06:
One nice suggestion I made for the FreeCad, but probably it will never be included, is to have a "Intersection Copy"...  it will be the same as "Intersection between two objects" but without deleting the objects at the end.   Suppose you have a cone, and you want to use a copy of the half tip of the cone to be used somewhere, you can produce any object large enough to gulf the half tip of the cone, a large cube for example, just positioning the cube in order for the cube to surround exactly the half of the cube you want to copy, and "Intersection Copy" will produce another object in the tree list, just the intersection, but without deleting either the original cone or the box.  So, you can select the new object (half tip of the cone) from the tree list and more wherever you need it.   Using a large sphere around the cone, will produce a copy of the half top of the cone, with a spherical base, etc.   This will be a neat addition to Boolean operations.  You think you can include it in your Design456 ?

MariwanJ, 2021-03-06:
Sorry, My example was bad .. I was trying to explain how you need to calculate the new position for the produced 2d object. i.e. you should decide where you put it .. and I didn't thought about that much.
Undo will works later but not now as I mentioned. 
Regarding your suggestion of new tool. Please make a official request and write down what you want, how it should work, and what is the expectation. I wish to hear that and I hope that I can make it some day. 
Not sure if "Discussion " is the write place but please put there .. I will look at it later where we keep these requests

closed 2021-03-06 by MariwanJ