MariwanJ/Design456 #15: Design456 Move Detail

wagnerlip, 2021-03-06:
Create a cube, click "Move Detail", a Transform window opens on Combo View, on Rotation, you can input any number in the "Around n-axis", click "Apply", the number returns to zero and the part doesn't rotate.

What happens is:  The "Rotation" pull down menu starts with "Rotation axis with angle" and followed by three number entires, for each angle, but those entries are for the "Euler angles (x y'z"), this is why it is not working.   I need to click the pull-down menu and selecte Euler angles, and back to "Rotation axis with angle" so it then shows only two entries, Axis and Angle, then it works.

Once in "Rotation axis with angle" it is expecting numbers on those entries (axis and angle), but the Euler entires is presented, so changing values on X, Y or Z entries will mean nothing to "Rotation axis with angle", then it does nothing to the object.

I think it is just an unmatch between the selection preset on the pulldown menu and the variables input selection.  Easy to fix.

wagnerlip, 2021-03-06:
Also, after you enter the correct angle to rotate, you see the object rotates nicely, but pressing Apply the rotated object jump to some odd position.  I could not create a relationship about the jumping distances, except it is larger if the angle rotated is also bigger.  Could not see the reason.

Also, on the same window, "Translation" and "Center" has the up/down icons at the right of each numeric entry, but they do nothing.  

At the bottom of the Translation and also Center entries, there are rectangles with text, the text doesn't fit into the rectangle and I can't understand what it means, the Translation appears as "ply ax" and the Center as "elected point".

I hope you don't get mad with me, but I am testing and pointing where to fix things, thank you for your job.

wagnerlip, 2021-03-06:
I also noted that, create a Design456 Cube, 10mm side standard, property view - placement - position it at 10,10,10 and call Design456 Move Detail.  On the transform screen that it opens, the Center comes automaticaly as the object (cube) position PLUS half the sizes, so X=15, Y=15 and Z = 15, to be exactly the center of the cube.

Two problems:
1) The little input field for Center X, Y and Z are not large enough and it DOES NOT show 15.000, instead it shows ".000", what confuses the user.
2) The rotation center point is then to be consider as to be 15,15,15, from the intersection axis (Grid) XYZ 0,0,0, but it is not, this 15.000 considers away from the center of the cube.  It rotates with a radius of 15.   To make it rotate around its own center, it needs to change the 15.000 to 0.000 in Center X, Y and Z.    The option "Center of Mass" doesn't make any difference if clicked or not, so I assume the "Center of Mass" is always active no matter the selection, this is why the 15,15,15 "from the center of mass" creates the large radius, and zero does not. 

It took me a while to understand why the cube was rotating very funny.   Or you use the coordinates XYZ from the axis intersection and option "center of mass" disable, or use 0,0,0 and "center of mass" enable, that is how it should work.

MariwanJ, 2021-03-06:
Regarding Move, you have 2 tools for that. Both are what you find in FreeCAD itself. I am just making it easy to call them. 
There might be some improvements, but I didn't put so much effort in testing them. So thanks you are testing them. 
I have to read carefully what you are referring to .. and maybe find a solution. I keep this open and come to it later. 
Don't hesitate to report any thing you find.. I just appreciate it. thanks.

wagnerlip, 2021-03-06:
Below the Combo View, Transform, Center, the 15.0000 that only fits ".0000" on the input field, I needed to select with mouse in order to see the integer part.   

And confirmed, the center of rotation is adding the "**Property View Position xyz**" plus the values on the "**Transform Center xyz**".   The "**Transform Center xyz**" is the same as "**Property View Position xyz**" + half of the "**Property View Length, Width and Height**", what is wrong.  

You must have one or another, not both, in this case the center of rotation becomes "**Position*2 + dimensions/2**"

[image]

wagnerlip, 2021-03-07:
I think the "center of mass" is wrong at the FreeCad software, a 10mm cube should have a center of mass of 5,5,5.   Even in FreeCad it is not, it adds the Position placement to this 5,5,5, and then rotates in real at 2xposition+5,5,5.  I already opened a discussion at FreeCad forum about it.   But your "use center of mass"  is not working at all.

MariwanJ, 2021-03-07:
Dear Wagnerlip,
My commands in Design456 is just calling the origional function. You can try it in the python console you will get the same results. What you find is due to how FreeCAD made these functions. I didn't implement them as I said, I just call them. 
Look at line 230 in my Design456_Aligenment.py   --> the command is ( Gui.runCommand('Std_TransformManip') 
Run that in the console you will get the same as if you right click on the object lists and you get the same command. 
That issue should be reported to FreeCAD forum .. I cannot change that it is in the core of FreeCAD. 
Thanks for reported and this will be closed.

closed 2021-03-07 by MariwanJ