MariwanJ/Design456 #16: Part Filet

wagnerlip, 2021-03-06:
Create a cube 10x10x10mm, select one edge and filet 1mm, select a connected edge and filet 1mm, it end up with 3 fileted edges.  It filet an extra edge without selecting it.

[image]

MariwanJ, 2021-03-06:
I will look at it later. Thanks for reporting

MariwanJ, 2021-03-07:
I tried the command on one edge of a cube, I couldn't reproduce the problem you have. 
Keep in mind that Fillet is dangours if you don't use it with caution you might ends up with invalid shape. 
Please would you try the fillet in the Part workbench? see if you can get the same issues. If the answer is YES, you should know that the issue is out of my control. Otherwise please update me. 
Thanks.

wagnerlip, 2021-03-08:
A cube generated in PARTS workbench, same size (10mm), same filleting sequence and dimension (1mm), does not present such double filleting issue.  I will keep testing.  

The only problem (I know) about filleting in FreeCad is when the filleting curve will not fit in the face, like a brick with 3mm wall and trying to filleting with 3.5mm, the object disappears.

MariwanJ, 2021-03-08:
Could you pleas look at the console and see what edge(s) are reported to get the fillet? You will see it  .
View->Panel->Python Console
Please keep in mind that you need to update your local repo in daily basis as I am updating in the same manner. So you get an updated version.

MariwanJ, 2021-08-25:
Fillet has many bugs. FreeCAD must fix them before anything I can do to it

closed 2021-08-25 by MariwanJ