MariwanJ/Design456 #23: Have you seen Defeaturing workbench

luzpaz, 2021-03-15:
https://github.com/easyw/Defeaturing_WB

https://wiki.freecadweb.org/Defeaturing_Workbench

Maybe there is functionality you can use in there for Design456 ?

MariwanJ, 2021-03-26:
I quickly looked at that WB. I will come back to look at it. At the moment I am busy with implementing the widget toolkit for coin3D. That will be the core of the new interface of visualization of manipulation of 3D shape..
I must get it work before I can do anything else. We had a discussion which way should I take and after that I decided to continue developing the widget set..It must make it easy to draw things in the 3D world much easier than now.
thanks for mentioning that.   I really wish to sum all the efforts in a GOOD workbench. I don't like the fragmentation we have in FreeCAD. Each part has a functionality which might be difficult for all users to know where and how to use it (install it).

MariwanJ, 2021-07-23:
@luzpaz @carlopav 
Hi, 
Even though many FreeCAD people know that I am working on the direct modeling project and I need to deeply understand the COIN3D and OpenCascade, no one really pointed out for me (for example) some sample codes that uses the script to draw inside coin3d world. 
One link that I share here is 
[https://github.com/coin3d/pivy/tree/master/examples/Mentor](url) 
which has got many Pivy sample images created by pivy or coin3d. That will definitely help in understanding how you can draw variant of shapes. 
There might be more out there .. which I don't know, but unfortunately no one tells you. Often I tried to find these kind of things in the documentation .. but no there isn't any hint there. 
It is sad that the time you need to learn becomes so long while you could make it shorter with help of the freecad community. 
Many time when I put even a simple question .. no one care .. Later I delete the question since there is no sense to keep it
just sad.

carlopav, 2021-07-23:
The link does not seem to work...

MariwanJ, 2021-07-23:
test it now

MariwanJ, 2021-07-23:
@luzpaz @carlopav 
Look at this new rep.
https://github.com/MariwanJ/COIN3D_Examples

carlopav, 2021-07-23:
Nice! @luzpaz would be good to add them to the Wiki if they are not already presente.

MariwanJ, 2021-07-24:
@luzpaz @carlopav 
All possible file-conversion are made. 
I really shocked how many info you get from these files. It will help me a to make nicer, better functionality in freecad .. I will be updating some of the new tools already made.. 
Using, reading the code help you alot.

luzpaz, 2021-07-24:
> would be good to add them to the Wiki if they are not already presents.

@carlopav any suggestions on where in the wiki ? (that's usually the hard thing to determine...lol)

MariwanJ, 2021-07-24:
@luzpaz 
 I believe it should be some where in the definition of Pivy and COIN3D .. But it should has it's own page. 
Later we can take screenshot of each example. 
This is a great opportunity to push up our development knowledge really.

luzpaz, 2021-07-24:
Any proposals for a page name?

MariwanJ, 2021-07-24:
Would you ask in the forum? :) maybe better to ask there .. 
The link is here . 
https://forum.freecadweb.org/viewtopic.php?f=10&t=60499&p=519132#p519132

carlopav, 2021-07-24:
> Any proposals for a page name?

I suggest https://wiki.freecadweb.org/Code_snippets
There Is a section with scene graph manipulation with pivy.

luzpaz, 2021-07-26:
created a placeholder https://wiki.freecadweb.org/Coin3d_snippets

MariwanJ, 2021-07-27:
@luzpaz 
please consider looking to the page I created .. and maybe release it 

https://wiki.freecadweb.org/Coin3d_snippets#Notes%3A
thanks.

MariwanJ, 2021-07-29:
@carlopav 
Have you seen that software? 
https://sourceforge.net/projects/coindesigner/#screenshots
http://coindesigner.sourceforge.net/

carlopav, 2021-07-29:
@MariwanJ interesting. I'll check It out!

MariwanJ, 2021-11-19:
I already looked at the de-featuring and I might use some of the code or ideas there. I close this as I am already looking at it and taking ideas from it. thanks.

closed 2021-11-19 by MariwanJ