MariwanJ/Design456 #24: [request] projecting a sketch on surface..

projectcad, 2021-03-24:
It needs also  a new feature  to project a sketch on a surface.

MariwanJ, 2021-03-26:
I believe that the functionality is there .. Draft can do that. Please look at the draft .. 
If not let me know.

projectcad, 2021-04-06:
Draft doesnt include this feature but you can adapt  the command "project to curved face" in Part WB to design456

Moreover, you can adapt the another command "facedraw"  of @microelly2 's Nurbs WB ( currently not active due to Python2) ...https://www.youtube.com/watch?v=bPLX3pwUoss&t=4

MariwanJ, 2021-04-06:
I like the idea. I thought Draft can attach the plane to the surface. 
I looked at https://github.com/microelly2/freecad-nurbs. I downloaded and try it . But none of the commands worked as they are written with older python as you said. I don't know the status of their project. I mailed the user .. let me see what he replies me.
Yes would be good to combine the tools he did. Many useful tools were there. 
You might know that I am working on the new interface for the direct modelling. I am quite busy, but your project can get priority. 
Thanks again.

projectcad, 2021-04-07:
@microelly2 had left his codes for uncertain time..you can change these codes as you want..due to GNU GPL licence..
[image]

MariwanJ, 2021-04-07:
Ok thanks. do you know if the liciens is like Freecad?

projectcad, 2021-04-07:
Absolutely like Freecad..... you can see licence terms at the beginning of @microelly2 's codes

MariwanJ, 2021-04-07:
Great. I started on looking on it. It is not made similar to the new WB. Might need reorganizing, more implementation might needed. It will take some time. I made a new branch just for that and will not be merged to either devbranch or main until it works.I I might update you in the future. Have a nice evening.

projectcad, 2021-04-08:
👍

MariwanJ, 2021-04-12:
@projectcad 
So sad to say that his work is just a puzzle. Many of his object are not encapsulated in classes which make it very difficult to find convert. 
He also made the interface quite complicated. 
I spent many many days in fining commands, and some other things he made. Unfortunately, he also uses external library which make it more difficult to use. Normal user will not be able to download. 
I didn't give up but it is quite challenging.
When I asked in the forum they answered me that he was more experimenting than writing usable code for the community. which is the fact that I see in his code. 
So sad that he spent so much time on his work but without resulting in a reusable work (without putting a lot of time to fix his work I mean).
Let me see what I can do. Still I am not able to fix at least one command of from his code.

MariwanJ, 2021-04-13:
I managed to get Design456 works again. There are alot of bug in the files but at least it can be run.

projectcad, 2021-04-17:
Great !..👍👍👍

MariwanJ, 2021-04-17:
Image to Cylinder is already done. You can prove it now.

projectcad, 2021-04-18:
But your code has not been updated since 19 days ago...

MariwanJ, 2021-04-18:
Don't look at the main branch .. 
Go to your director MOD under user account. When you download with git command .. then switch to the branch for the nurbs like bellow 
```
git clone https://github.com/MariwanJ/Design456
git checkout nurbs 

```
and start FreeCAD

projectcad, 2021-04-18:
At first attempt it returns an error...
Just then  reattempting returns the broken WB.. and then Freecad suddenly closes while doing something in this broken WB..
[image]
[image]

projectcad, 2021-04-18:
Ok..it seemed after installing "scipy" Python module ...currently i am trying commands..
[image]

MariwanJ, 2021-04-18:
There are several libraries you need to install before you can run the Nurbs. 
It is not my WB which has problem .. It is his code .. Rather difficult to deal with .. No clear vision how it works, how it organized. 
But I hope I fix the problem soon. I will merge it to the develop as soon as I rearrange the icons. It is mess at the moment.

MariwanJ, 2021-04-18:
http://freecadbuch.de/doku.php?id=nurbs

projectcad, 2021-04-18:
I tried imagesurface... but nothing seems..
[image]

projectcad, 2021-04-18:
Maybe one missing point there... @microelly2 uses nodeeditor with pyflow in his Nurbs example videos..

projectcad, 2021-04-18:
Maybe one missing point there... 
@microelly2 uses nodeeditor-pyflow in his Nurbs example videos..

MariwanJ, 2021-04-18:
The exaples he is giving (file names) might not exist .. update them .. use your own. try with small picture (48x48) .
Notice that there are still commands not included. 
I will merge today the nurbs to devbranch .. and I might merge it to the main today. let me see.

MariwanJ, 2021-04-20:
A big push to the rep. Still I didn't merge it to the main as I have some work to do .. but toolbars are almost fixed. 
Notice that there are many many bugs still. I just want to list the commands and make class for them .. after that I will start testing them and finding the problems.

MariwanJ, 2021-04-20:
> itor-pyflow
Pyflow ? I don't know if I will touch it.. I need to go back to my main goal. I don't know if people will/would ever use it.

projectcad, 2021-04-22:
Just looked at  Nurbs adapted to Design456 bu you, many commands  has started to active after succesfull conversion of codes from Python2 to Python3..in spite of some missing points.. i think..these codes also would be benefit for your actual target in Design456  @luzpaz @galou @nighcoder

MariwanJ, 2021-04-22:
At least I learn something :) .. I will try my best to eliminate all bugs. 
Microelly in his code never check if you are selecting objects before clicking on the command. He assues that you do that. But that is a bad decision. That is why I need to implement that for the whole code. Too many hours will be spent only on that. 
I feel sad that he can so much but, he is not writing good code unfortunatlly.. No comments, no HOWTO use the tools. It is just nightmare. 
I hope in the future, I further develop them also. But at the moment I want them to be there .. 
Thanks for you suggestions and openion. BR

MariwanJ, 2021-04-25:
Due to a big discouragement from the FreeCAD forum, Nurbs is separated and I will stop further developing it at least for now. It has it's own WB now.
please refer to https://github.com/MariwanJ/nurbs

MariwanJ, 2021-05-11:
@projectcad
> 
> 
> 👍👏👏👏Just looked at Nurbs adapted to Design456 bu you, many commands has started to active after succesfull conversion of codes from Python2 to Python3..in spite of some missing points.. i think..these codes also would be benefit for your actual target in Design456 @luzpaz @tomate44 @galou @nighcoder

I managed to fix the "Draw on Face" command. 
It is published and you can use it .. 
Please look at this video (not so good as my computer cannot capture dialogs )
[https://youtu.be/NMGpnWF-x3A](url)

projectcad, 2021-05-12:
Just updated and tried but mine returns still same error..
[image]

MariwanJ, 2021-05-12:
@ projectcad 
It seems that you didn't update your version. Please remove it and make a new clone. 
I just used my version and it works.

projectcad, 2021-05-12:
Deleted Nurbs and reinstalled..but changed error 's line this time..🙄
[image]

MariwanJ, 2021-05-12:
Why do the cube is like so? Did you do anything to the cube?

MariwanJ, 2021-05-12:
I updated the method to generate an error if you don't select a face. It should work ..

projectcad, 2021-05-12:
I selected face of cube in 
[image]
example above..
And also tried like that..

MariwanJ, 2021-05-12:
I removed my local copy, downloaded the github copy and it is working. Which command you are using? 
Nurbs_FaceDraw should be used.
please make a gain a pull to update your version.  remove the ( __pycache__ ) folder. And start again FreeCAD. 
I cannot reproduce the error you have . Put your version of FreeCAD also please

projectcad, 2021-05-12:
Yes "Nurbs_FaceDraw"...like in photos..
I deleted Nurbs then reinstalled with git clone again...tried again in both version.. Freecad Linkstage3 git version and Freecad Daily..but...same result..only one difference is line of the error
[image]
[image]

MariwanJ, 2021-05-12:
Please update and try it again. I changed something might be the cause. 
I will let you know later if it works.

projectcad, 2021-05-12:
also please find below..

OS: Linux Mint 19.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.20.
Build type: Release
Branch: unknown
Hash: 28c51d4a43ce...............
Python version: 3.6.9
Qt version: 5.9.5
Coin version: 4.0.0a
OCC version: 7.5.1
Locale: English/UnitedStates (en_US)

OS: Linux Mint 19.3 (X-Cinnamon/cinnamon)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.27812 (Git)
Build type: Unknown
Branch: LinkStage3
Hash: b8cd87c34a66..............
Python version: 3.6.9
Qt version: 5.9.5
Coin version: 4.0.0a
OCC version: 7.5.1
Locale: English/UnitedStates (en_US)

MariwanJ, 2021-05-12:
Ok. Strangely It was working for me. But there are many many bugs in that file. Don't know how it was working for me before. 
It is the way he wrote his files. Quite unstructured and complex to followup. 
I will let you know when I fix that.

MariwanJ, 2021-05-14:
@projectcad 
Now, I tried 2 days to resolve the bugs that Nurbs had in the facedraw.. It is endless. 
I couldn't fix it. I don't know how it was working for me before really. I am giving up for now.. I need to take a break and come back to that another time.. Frustrating!

closed 2021-05-14 by MariwanJ