MariwanJ/Design456 #25: Suggestion on how to implement the interactive interface for the Direct modelling

MariwanJ, 2021-03-27:
https://forum.freecadweb.org/viewtopic.php?f=10&t=57032&p=491667#p491667: 

Re: New Widget system for 3DCoin (Aimed to be used with direct modeling tools)

    Report this post
    Quote

Postby kwahoo » Sat Mar 27, 2021 8:32 pm

    marja1 wrote: ↑
    Fri Mar 26, 2021 2:49 pm
    I choose coin3d. I will try my best to find the easiest way to implement the desired functionality.
    You need sometime to understand how you do things.. 

I do some similar tinkering for VR/XR interaction. The basic idea is:
a) build a separate Coin3D scenegraph just for menu items,
b) do picking action (SoRayPickingAction) for "menu scenegraph", find what menu item was hit
c) if nothing was hit, perform picking action for "world" scenegraph
d) check what was hit, use SoDetail to find which part of shape was hit, find normal face (that will be useful for rendering manipulators)
e) find corresponding TopoShape using getViewProviderByPath (similar to this https://forum.freecadweb.org/viewtopic.php?t=25355 )

MariwanJ, 2021-03-27:
https://forum.freecadweb.org/viewtopic.php?t=25355

MariwanJ, 2021-03-27:
Could this help in getting the TOP/BOTTOM/FRONT/BACK toolbar on the scene ?
```

from PySide2 import QtWidgets

def findInChildren(obj, searched):
	for child in obj.children():
		if isinstance(child, searched):
			return child
		else: 
			res = findInChildren(child, searched)
			if res:
				return res

	return None

view = findInChildren(Gui.getMainWindow(), QtWidgets.QGraphicsView)
widget = QtWidgets.QLineEdit()

proxy = view.scene().addWidget(widget)
```

MariwanJ, 2021-03-28:
```
from pivy import coin

def clic_cb( event_callback):
    event = event_callback.getEvent()
    if ( type(event) == coin.SoMouseButtonEvent and
            event.getState() == coin.SoMouseButtonEvent.DOWN and
            event.getButton() == event.BUTTON1):
        print("Left mouse button pressed")
        render_manager = view.getViewer().getSoRenderManager()
        ray_pick = coin.SoRayPickAction(render_manager.getViewportRegion())
        ray_pick.setPoint(coin.SbVec2s(*event.getPosition()))
        ray_pick.apply(render_manager.getSceneGraph())
        picked_points = ray_pick.getPickedPointList()
        for pp in picked_points:
            path = pp.getPath()
            length = path.getLength()
            point = path.getNode(length - 1)
            print('%s'%str(point))

view = FreeCADGui.ActiveDocument.ActiveView

clicCB = view.addEventCallbackPivy( coin.SoMouseButtonEvent.getClassTypeId(), clic_cb)

#view.removeEventCallbackPivy( coin.SoMouseButtonEvent.getClassTypeId(), clicCB)

```

MariwanJ, 2021-03-29:
https://www.iesensor.com/download/FreeCAD_Mod_Dev_Guide__20160920.pdf

MariwanJ, 2021-04-07:
from pivy import coin
import FreeCADGui as Gui

_group = coin.SoGroup()
_font = coin.SoFont()
_text = coin.SoText2()

_text.string.setValue('HELLO!')
_font.name = 'Arial'
_font.size.setValue(200.0)

_group.addChild(_font)
_group.addChild(_text)

Gui.ActiveDocument.ActiveView.getSceneGraph().insertChild(_group, 0)

MariwanJ, 2021-04-12:
look at here 

class WorkSpace():

	def __init__(self, name):
		try:lidok= App.getDocument(name)
		except:	lidok=App.newDocument(name)
		self.dok=lidok
		self.name=name

	def delete(self):
		App.closeDocument(self.name)

	def addObject2(self,obj,count=10):
		if 0:
			res=self.dok.addObject("Part::FeaturePython",obj.Name)
			ViewProvider(res.ViewObject)
		#return
		res=self.dok.addObject("Part::Spline",obj.Name)
		res.Shape=obj.Shape.toNurbs()
		#return
		f=obj.Shape.Face1.Surface
		cs=[]

		for ui in range(count+1):
				cs.append(f.uIso(1.0/count*ui).toShape())
		for vi in range(count+1):
				cs.append(f.vIso(1./count*vi).toShape())
		res.Shape=Part.Compound(cs)

	def recompute(self):
		self.dok.recompute()

	def show(self):
		self.getWidget().show()

	def hide(self):
		self.getWidget().hide()

	def getWidget(self):
		mw=FreeCADGui.getMainWindow()
		mdiarea=mw.findChild(QtGui.QMdiArea)

		sws=mdiarea.subWindowList()
		print "windows ..."
		for w2 in sws:
			print str(w2.windowTitle())
			s=str(w2.windowTitle())
			if s == self.name + '1 : 1[*]':
				print "gefundne"
				return w2
		print self.name + '1:1[*]'

	def _haha(self):
		pass

MariwanJ, 2021-04-22:
color = coin.SoBaseColor()
font = coin.SoFont()
font3d = coin.SoFont()
text = coin.SoAsciiText() 
text3d = coin.SoText2()  
text.string="test"
coin.SoTransform()
coin.SoText2.CENTER #2d
coin.SoText2.LEFT  #2D
coin.SoAsciiText.RIGHT #3d
coin.SoAsciiText.CENTER #3d
#text size
region = view.getViewer().getSoRenderManager().getViewportRegion()
action = coin.SoGetBoundingBoxAction(region)
text.getBoundingBox(action)
return action.getBoundingBox().getSize().getValue()

MariwanJ, 2021-04-22:
def getEditNode(self, pos):
        """Get edit node from given screen position."""
        node = self.sendRay(pos)
        return node
    
    def sendRay(self, mouse_pos):
        """Send a ray through the scene and return the nearest entity."""
        ray_pick = coin.SoRayPickAction(self.render_manager.getViewportRegion())
        ray_pick.setPoint(coin.SbVec2s(*mouse_pos))
        ray_pick.setRadius(self.pick_radius)
        ray_pick.setPickAll(True)
        ray_pick.apply(self.render_manager.getSceneGraph())
        picked_point = ray_pick.getPickedPointList()
        return self.searchEditNode(picked_point)

    def searchEditNode(self, picked_point):
        """Search edit node inside picked point list and return node number."""
        for point in picked_point:
            path = point.getPath()
            length = path.getLength()
            point = path.getNode(length - 2)
            #import DraftTrackers
            if hasattr(point,"subElementName") and 'EditNode' in str(point.subElementName.getValue()):
                return point
        return None

MariwanJ, 2021-04-22:
self.trans.translation.setValue([position.x, position.y, position.z])

MariwanJ, 2021-04-26:
import fr_coinwindow as wn
import fr_line_widget as line
import FreeCAD as App
g=[]
p1=(0,0,0)     # first point in the line and for the windows
g=[]
p2=(400,400,0) # width and height of the window .. z is not a matter (generally this has no effect)
p3=(20,20,0)   # Second point in the line 
g.append(p1)
g.append(p2)
wny=wn.Fr_CoinWindow(g,"MyWindow")  # Create the window, label has no effect at the moment
g.clear()
g.append(p1)
g.append(p3)
ln =line.Fr_Line_Widget(g,"my wline",20)   # draw the line - nothing will be visible yet
wny.addWidget(ln)              # Add it to the window as a child 
wny.show()                    # show the window and it's widgets.

MariwanJ, 2021-04-26:
pos=App.Vector(30,30,0)
color=(1,1,1)
text=["aaa","bbb"]
textpos = coin.SoTransform()
p=pos
textpos.translation.setValue(p.x+10,p.y+10,p.z+10)
font = coin.SoFont()
font.size = 20
#text2d = coin.SoText2()
text3d = coin.SoAsciiText()

#text2d.string.setValues([l.encode("utf8") for l in text if l])
text3d.string.setValues([l.encode("utf8") for l in text if l])

myMaterial = coin.SoMaterial()
myMaterial.diffuseColor.set1Value(0, coin.SbColor(*color))

sg    =  coin.SoSeparator()
#self._sg= sg
root=FreeCADGui.ActiveDocument.ActiveView.getSceneGraph()
root.addChild(sg)
#    
#    
node2d    =  coin.SoSeparator()
node2d.addChild(textpos)
node2d.addChild(myMaterial)
node2d.addChild(font)
#node2d.addChild(text2d)
node2d.addChild(text3d)

sg.addChild(node2d)

MariwanJ, 2021-04-27:
def move(self, delta):
        """Move the ghost to a given position.

        Relative from its start position.
        """
        self.trans.translation.setValue([delta.x, delta.y, delta.z])

    def rotate(self, axis, angle):
        """Rotate the ghost of a given angle."""
        self.trans.rotation.setValue(coin.SbVec3f(DraftVecUtils.tup(axis)), angle)

MariwanJ, 2021-04-28:
textdrawstyle = coin.SoDrawStyle()
        textdrawstyle.style = coin.SoDrawStyle.FILLED

        # The text string needs to be initialized to something,
        # otherwise it may crash
        self.text2d.string = self.text3d.string = "Label"
        self.text2d.justification = coin.SoText2.LEFT
        self.text3d.justification = coin.SoAsciiText.LEFT

        self.node2d = coin.SoGroup()
        self.node2d.addChild(self.trans)
        self.node2d.addChild(self.mattext)
        self.node2d.addChild(textdrawstyle)
        self.node2d.addChild(self.font)
        self.node2d.addChild(self.text2d)

        self.node3d = coin.SoGroup()
        self.node3d.addChild(self.trans)
        self.node3d.addChild(self.mattext)
        self.node3d.addChild(textdrawstyle)
        self.node3d.addChild(self.font)
        self.node3d.addChild(self.text3d)

MariwanJ, 2021-05-10:
https://github.com/FreeCAD/FreeCAD/blob/master/src/Mod/TemplatePyMod/FeaturePython.py#L192

MariwanJ, 2021-05-17:
import numpy as np
from pivy import coin

from freecad import app
import FreeCADGui as gui

app.newDocument()
sg = gui.activeDocument().activeView().getSceneGraph()

# create a degree 4 bspline with same weights:

array = np.array(
  [[0.  , 0.  , 0.01, 0.07, 0.18, 0.36, 0.5 , 0.71, 1.  ],
   [0.  , 0.02, 0.05, 0.09, 0.1 , 0.08, 0.06, 0.04, 0.  ],
   [0.  , 0.  , 0.  , 0.  , 0.  , 0.  , 0.  , 0.  , 0.  ],
   [1.  , 1.  , 1.  , 1.  , 1.  , 1.  , 1.  , 1.  , 1.  ]])

# The knot vector
knots = ([0] * 5 + [1] * 2 + [2] *2 + [3] * 5)

curveSep = coin.SoSeparator()
complexity = coin.SoComplexity()
controlPts = coin.SoCoordinate4()
curve      = coin.SoNurbsCurve()
controlPts.point.setValues(0, array.shape[1], array.T.tolist())
curve.numControlPoints = array.shape[1]
curve.knotVector.setValues(0, len(knots), knots)
curveSep += [complexity, controlPts, curve]
sg += curveSep

MariwanJ, 2021-05-17:
from pivy import coin
sg = Gui.ActiveDocument.ActiveView.getSceneGraph()
svg = coin.SoImage()
svg.filename = 'routeFilename.svg'
imagePos = coin.SoTransform()
imagePos.translation.setValue([10,0,0])
imagePos.rotation = coin.SbRotation(0,0,0,0)
image = coin.SoSeparator()
image.addChild(svg)
image.addChild(imagePos)
sg.addChild(image)
--------------------------------------------------------

from pivy import coin
sg = Gui.ActiveDocument.ActiveView.getSceneGraph()
svg = coin.SoTexture2()
svg.filename = '/Users/juanvanyo/Desktop/FreeCAD/FreeCAD-GDT/Gui/Resources/icons/Characteristic/perpendicularity.svg'
box = coin.SoVRMLBox()
box.size = (2,2,0)
imagePos = coin.SoTransform()
imagePos.translation.setValue([10,0,0])
imagePos.rotation = coin.SbRotation(0,0,0,0)
image = coin.SoSeparator()
image.addChild(imagePos)
image.addChild(svg)
image.addChild(box)
sg.addChild(image)

MariwanJ, 2021-05-17:
line = graphics.Line([[0, 0, 0], [1, 0, 0], [2, 0, 0]])
sg += line
line.data.point[1].setValue([1, 1, 0])

MariwanJ, 2021-05-20:
#Inventor V2.1 ascii
Coordinate4 {
   point [
     0.0 -5.0     0.0 1.0,
     2.5 -2.5     0.0 0.5,
     2.5 -0.66987 0.0 1.0,
     0.0  1.94013 0.0 0.5,
    -2.5 -0.66987 0.0 1.0,
    -2.5 -2.5     0.0 0.5,
     0.0 -5.0     0.0 1.0
   ]
}
NurbsCurve {
   numControlPoints 7
   knotVector [ 0, 0, 0, 1, 1, 2, 2, 3, 3, 3 ]
}
# debug: show all the control points
BaseColor { rgb 1 1 0 }
DrawStyle { pointSize 3 }
PointSet { numPoints 7 }

luzpaz, 2021-05-20:
Consider using the ``` markdown for code 
```
Code goes in here
   indents are represented correctly
```

MariwanJ, 2021-05-23:
These are notes to myself. Nothing for public sorry. But since I work on different computers, I have no other /better way to save notes to myself. 
:)
They will be deleted as soon as I use them in the project.

luzpaz, 2021-05-30:
That's cool. All I'm saying is that code markdown respects indentation. So it's easier to copy/paste. Easier for you

MariwanJ, 2021-05-30:
> 
> 
> That's cool. All I'm saying is that code markdown respects indentation. So it's easier to copy/paste. Easier for you

Ok I thought you mean for others. Not a big deal .. but i did that .. thanks anyway

MariwanJ, 2021-06-25:
[https://gist.github.com/yorikvanhavre/5953106](url)

MariwanJ, 2021-07-21:
from pivy import coin

pts = [FreeCAD.Vector (3.47, 1.19, -1.24), FreeCAD.Vector (3.51, 2.46, -1.38), FreeCAD.Vector (3.17, 3.73, 1.37), FreeCAD.Vector (3.23, 4.93, 0.59),
       FreeCAD.Vector (2.02, -0.12, -0.32), FreeCAD.Vector (2.02, 2.56, 1.04), FreeCAD.Vector (2.12, 4.05, 1.69), FreeCAD.Vector (2.19, 5.26, 0.91),
       FreeCAD.Vector (0.31, 0.59, -0.50), FreeCAD.Vector (1.05, 3.45, 1.50), FreeCAD.Vector (0.64, 3.99, 3.18), FreeCAD.Vector (0.67, 5.15, 2.374),
       FreeCAD.Vector (-0.51, 1.18, 0.59), FreeCAD.Vector (-0.40, 2.78, 1.85), FreeCAD.Vector (-0.51, 3.68, 0.59), FreeCAD.Vector (-0.48, 4.84, -0.21)]
Uknots = [0.0]*4 + [1.0]*4
Vknots = [0.0]*4 + [1.0]*4

material = coin.SoMaterial()
material.transparency.setValue(0.0)
material.diffuseColor.setValue(coin.SbColor(0.8,1,0.8))
material.specularColor.setValue(coin.SbColor(1,1,1))
material.shininess.setValue(1.0)

controlPts = coin.SoCoordinate3()
controlPts.point.setValues(0, len(pts), pts)

surface    = coin.SoNurbsSurface()
surface.numUControlPoints = 4
surface.numVControlPoints = 4
surface.uKnotVector.setValues(0, len(Uknots), Uknots)
surface.vKnotVector.setValues(0, len(Vknots), Vknots)

# prop = coin.SoNurbsProperty() #word censored! Not available in coin3d

surfaceNode = coin.SoSeparator()

surfaceNode.addChild(controlPts)
surfaceNode.addChild(material)
surfaceNode.addChild(surface)

view = Gui.ActiveDocument.ActiveView
sg = view.getSceneGraph()
sg.addChild(surfaceNode)

#material.transparency.setValue(0.5) ### this line will crash FC

MariwanJ, 2021-07-23:
https://www.dgp.toronto.edu/public_user/stam/reality/Research/pdf/qtEG.pdf

MariwanJ, 2021-07-23:
_
In FreeCAD we do not use any of the Coin3d solid objects (cube, cone, sphere, cylinder) instead we use an SoIndexedFaceSet (or better a derived class of it SoBrepFaceSet). This way you can handle any kind of solids.
You basically need two nodes: an SoCoordinate3 node to hold all the 3d coordinate points and the SoIndexedFaceSet (or alternatively an SoFaceSet) node which holds the indexes for the triangles.

If you have a mathematical description of a solid you must run the tessellation to get the above information. In FreeCAD we use OCCT's capabilities therefore.
_

https://forum.freecadweb.org/viewtopic.php?t=34795

MariwanJ, 2021-07-23:
https://wiki.freecadweb.org/Scripted_objects

MariwanJ, 2021-07-23:
https://github.com/coin3d/pivy/tree/master/examples/Mentor

MariwanJ, 2021-08-07:
[https://forum.freecadweb.org/viewtopic.php?p=521318#p521318](url)

MariwanJ, 2021-08-15:
@luzpaz 
I started to write some pages on the Fr_Widget system. I want to make tutorials, and explanation and documentation at the wiki pages. Please would you give me your opinion if the pages are ok or what I should do? 
I didn't want to waste my time in writing in the forum.. Better to once write a proper documentation for the widgets system. and by that people will find it and I hope they will come to help :)
https://wiki.freecadweb.org/FreeCAD_GUI_Widgets_toolkit
I even added my workbench to external workbench page. See if I didn't make any mistake. 
Thanks in advance.

MariwanJ, 2021-11-08:
https://documentation.help/Open-Cascade/occt_user_guides__shape_healing.html#occt_shg_5_

MariwanJ, 2022-01-04:
[https://wiki.freecadweb.org/index.php?title=Mesh_to_Part](url)

MariwanJ, 2022-01-11:
`from pivy import coin

##### Create the static text node

sep = coin.SoSeparator()

cam = coin.SoOrthographicCamera()
cam.aspectRatio = 1
cam.viewportMapping = coin.SoCamera.LEAVE_ALONE

font = coin.SoFont()
font.name = "Arial"
font.size = 14

text = coin.SoText2()
text.string.setValues(0,2,["This is a static 2D text","in the 3D view"])

trans = coin.SoTranslation()
trans.translation = (-0.95,0.95,0)

color = coin.SoBaseColor()
color.rgb = (0,0,0)

sep.addChild(cam)
sep.addChild(trans)
sep.addChild(color)
sep.addChild(font)
sep.addChild(text)

##### Add the node to the renderManager as a Superimposition

view = FreeCADGui.ActiveDocument.ActiveView
viewer = view.getViewer()
render = viewer.getSoRenderManager()
sup = render.addSuperimposition(sep)

sg=FreeCADGui.ActiveDocument.ActiveView.getSceneGraph()
sg.touch()
`

MariwanJ, 2022-01-28:
OCC Api

`aReshape.Remove(myAISContext->SelectedShape());
Handle_AIS_InteractiveObject picked;
picked= myAISContext->SelectedInteractive();
Handle(AIS_Shape) aShape=Handle(AIS_Shape)::DownCast(picked);
TopoDS_Shape aNewShape=aReshape.Apply(aShape->Shape());
aShape->Set(aNewShape);
myAISContext->Redisplay(aShape);`

MariwanJ, 2022-02-17:
http://www.songho.ca/opengl/gl_sphere.html

MariwanJ, 2022-11-23:
opengl & OCCT
https://unlimited3d.wordpress.com/2021/07/29/occt-viewer-and-gtk/
https://dev.opencascade.org/content/how-use-opengl-directly-occ%EF%BC%9F

MariwanJ, 2023-04-18:
https://github.com/Wanna-Get-High/M3DS

closed 2023-12-15 by MariwanJ