MariwanJ/Design456 #30: 3D Snap Suggestion

projectcad, 2021-05-18:
It is very important point that the feature "3D Snap" is still missing in FreeCAD...if it becomes, reference line among 3D objects for a Datum Plane becomes very easy ...and  more...
@yorikvanhavre @luzpaz @realthunder

MariwanJ, 2021-05-19:
Could you find any thing like that in Nurbs? I think there might be .. so I can take it and put it in my WB. 
I am busy these days and might not have time to look at it .. I appreciate that.

MariwanJ, 2021-05-20:
The way I think about snap in general is not how Freecad draft or sketcher works. Rather than that, I will prefer giving the dimensions of the sketch either by entering during creation of the object or using smart scale. 
I am trying to implement at the moment smart scale. 
Another thing which is important, is to be able to draw on any face without having to click several commands. It should just automatically be possible.. how? needs some thinking. 
I think the commands are there but the interface .. no .

realthunder, 2021-05-21:
Does the auto scale below fits the 'smart scale' function you have in mind? It is implemented in [this](https://github.com/realthunder/FreeCAD/commit/ab88228d823feab9ca7e4fab19ff83b8c197e3bf) commit in my branch. Not sure if it is usable to you.

[image]

MariwanJ, 2021-05-23:
smart scale is a way to resize objects. It has nothing to do with snapping or GRID. The term is used by 123D Design. 
[https://youtu.be/w5_MLVb3YmY?t=61](url)

projectcad, 2021-05-26:
I just was aware of there is already 3D Snap..but if "Relative" box is unchecked in commands "Line", "Wire", "Bspline"....

closed 2021-11-19 by MariwanJ