MariwanJ/Design456 #31: Inheretance callbacks dosen't work. Theoretically it should work. please help!

MariwanJ, 2021-06-07:
@luzpaz @carlopav @joelgraff
For the widget system I am preparing,  I let the user write the callback for the mouse events later by sub-classing the widgets. 
But when I subclass the line_widget in the smart-scalse, the callback is still the parents callback. There is no reason to be like that as I am overriding the variable. 
If it was cpp, it shouldn't happen but in python either I don't know or it is not easy to implement. 
I would appreciate any help in understanding or fixing this issue.  Thank you very much in advance.
look at this part. 
`
def lblcallback(userData=None):
    """
            This function will run the label-changed 
            event callback.
    """
        #TODO : Subclass this and impalement the callback 
        #          to get the desired effect
    print("dummy line-widget-label callback")
             
def callback(userData=None):
    """
            This function will run the when the line is clicked 
            event callback. 
    """
        #TODO : Subclass this and impalement the callback 
        #          to get the desired effect
    print("dummy line-widget callback" )

class Fr_Line_Widget(fr_widget.Fr_Widget):

    """
    This class is for drawing a line in coin3D world
    """

    def __init__(self, vectors: List[App.Vector] = [], label: str = "", lineWidth=1):
        self.w_lineWidth = lineWidth  # Default line width
        self.w_widgetType = constant.FR_WidgetType.FR_EDGE
        
        self.w_callback_=callback           #External function                      <---- this is not overriding the function
        self.w_lbl_calback_=lblcallback     #External function                    <---- this is not overriding the function
        self.w_KB_callback_=KBcallback      #External function
        self.w_move_callback_=movecallback  #External function
        super().__init__(vectors, label)
`

luzpaz, 2021-06-07:
@realthunder can you take a quick look at this?

MariwanJ, 2021-06-07:
@luzpaz
I believe the solution is as I asked at stackoverflow. 
[(https://stackoverflow.com/questions/67877603/how-to-override-a-function-in-an-inheritance-hierarchy#67877671)]
I will try to do the modifications required as per the solution they told me there.