MariwanJ/Design456 #60: Update to new FreeCAD AddonManager metadata schema

luzpaz, 2022-01-19:
See https://github.com/FreeCAD/FreeCAD-addons/issues/200

MariwanJ, 2022-01-20:
@luzpaz 
Please see if that is what was required. 
I copied from another WB.
Regarding the version, still I have no version since I didn't make a complete tool .. And I wouldn't name  the current status any version number.
Let me know if that is Ok please. 
Thanks for the tips.

MariwanJ, 2022-01-20:
@luzpaz 
With the file I added based on the recommendations, freecad dosen't start at all. removing the file , freecad works again. 
so don't know what is the problem. look at the file at my github.

luzpaz, 2022-01-20:
https://github.com/MariwanJ/Design456/blob/9fc98aa829dc171ba86d244341040f41bc08a842/package.xml#L21-L23

Remove these tags and just add this tag for now:  
`<tag>direct-modeling</tag>`

MariwanJ, 2022-01-20:
@luzpaz 
I did that but I can no longer see my workbench in freecad.
I just downloaded latest weekly build freecad 
`OS: Windows 10 (10.0)
Word size of FreeCAD: 64-bit
Version: 0.20.26761 (Git)
Build type: Release
Python version: 3.8.12
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.5.3

`

MariwanJ, 2022-01-20:
Definitely, having that file, doesn't allow freecad to load the wokbench name .. You will no longer be able to run the workbench.
I will removed from main so people not become affected.

closed 2022-01-21 by MariwanJ