MariwanJ/Design456 #63: Clarifying between "interactive" and "direct modelling"

yashpalgoyal1304, 2022-01-19:
suggestion:
* also, direct modelling can be precise as well, that is, can be done with numeric input too. what more, i think that is crucial to have too, 
* so, hybrid approach. like the transformation widgets, and input box too
* the input box can be made available on canvas too as kinda like a stretch goal
* i have tried to make readme more apt in this context at 
https://github.com/MariwanJ/Design456/pull/61/files#diff-b335630551682c19a781afebcf4d07bf978fb1f8ac04c6bf87428ed5106870f5R10

> For example, clicking on a face and being able to extrude by moving the mouse, instead of specifying a the distance by entering a number in to a text field.
> _- https://github.com/MariwanJ/Design456/blob/main/README.md?plain=1#L6-L8_

MariwanJ, 2022-01-20:
@yashpalgoyal1304 
There is a big issue in FreeCAD. I explained that several time on the freeCAD forum and on youtube.
 FreeCAD is slow. 
Direct manipulation is not possible without simulating the whole thing in another drawing library. COIN3D is not good. And I am still thinking how I can achieve a better interactive feeling of the whole WB. 
I don't know if Vulkan is added to FreeCAD, I don't know either much about Vulkan.
I get almost no help from FreeCAD forum rather than that I get disrespect. So I left there.
Now, I keep my Workbench away from freecad other wb (sketcher specially, and other WB). 
My Road map is to try to achieve the functionalities I know and see how FreeCAD can give me API. After knowing the limitation and the possibilities, I make improvements.
At the moment, I need tools for joining, simplifying, repairing, ..etc all 2D shapes (line, face, circle, curved faces ..etc). Still it is difficult to know how you can do some tools I made before for curved surfaces. 

Now, you say (i think you have mingled direct modelling with interactive modelling), to be honest I don't see any different .. but please let me know how I can or why I should distinguish between them? 
Now, having it with dialogs is preferred for sure. But at this stage, I wouldn't do that as it make my code more complex while I am not trying to get the best UI. 
You can see that I wrote a TODO for fixing the interface. But that is not done before I complete the manipulations tools. 
Please notice that I am working along on this project. My knowledge might be limited .. I am learning!!

I close this but I appreciate your notes. I might reopen this in the future. 
Thanks.

closed 2022-01-20 by MariwanJ

yashpalgoyal1304, 2022-01-20:
> I get almost no help from FreeCAD forum rather than that I get disrespect

uh oh, that's sad to hear. seems like no project is perfect. anyways....

> how I can (learn to) distinguish between them

as i hinted about it in my MR as well:
* direct modelling means you make changes directly to the geometry.  \
the parameters or say feature timeline is NOT preserved, so, no hinderance there
* while "interactive" means that you can do the action you want using some thing which gives you direct feedback \
so, the interactive UI widgets

+ these two are separate things
+ you can have direct modelling without interactive widgets, and similarly
+ you can have interactive widgets in non-direct modelling as well

> why I should distinguish between them

to save time, and being able to focus on getting the core functionality there, without having to worry about getting the widgets done right

> I wouldn't do that (floating number field dialogs) as it make my code more complex while I am not trying to get the best UI

yeah, i completely understand that. i am not saying to have dialogs now. i just shared that as a thought for future (hence the term before them: stretch goal)

MariwanJ, 2022-01-20:
One of the biggest problem you get when you try direct model any object in freecad, most of the time you have to re-create the object to get your code works. 
Take the "Reset Placement" I did for some days ago. You will never, ever be able to reset the object without recreating it. The problem is in the OCC. Means that whenever an object creates, it is read-only and you are not able to change anything of it.
That pushes me often to make a simple-copy of the old object and work on it. 
Compound objects are a huge black box. None of the regular code work on them. They are not an object rather than a container of objects which you don't get many things out of the box for them (normalAt, Direction..etc).
So, you have to convert it to a simple copy to work on it. 
I am afraid as I said, I wouldn't keep the history of the objects at the end. Which means it is a direct modeling workbench. 
I don't like the idea either to have so many inheritance of object. You can easily break the whole chain by a small mistake. That is the nature of what you find in sketcher WB.

yashpalgoyal1304, 2022-01-20:
yeah, i totally understand that. i am just saying that you can skip implementing the interactive widgets for now