MariwanJ/Design456 #78: Update images at the wiki pages?

MariwanJ, 2022-06-30:
@luzpaz 
HI, 
Would you please remind me how I can upload images to the wiki pages? 
What was the link? 
I updated some of the tool-lists and needs to update the images. 
Thanks,
Mariwan

luzpaz, 2022-06-30:
@MariwanJ Check out https://wiki.freecad.org/WikiPages#Graphics

luzpaz, 2022-06-30:
Can you in general log in to the wiki ?

MariwanJ, 2022-06-30:
@luzpaz 
My username was marja1
I was in the early morning, and made some changes to the wiki but not today.  I cannot relogin anymore.

luzpaz, 2022-06-30:
It doesn't look like you were blocked at all. Can you log in to wiki.freecadweb.org instead? (sometimes the domain name is the issue)

MariwanJ, 2022-06-30:
@luzpaz 
You are correct. but still I am not able to login to https://wiki.freecad.org/Special:Upload.
It asks again for login. 
don't know why

luzpaz, 2022-06-30:
Show me a screenshot of you logged in

MariwanJ, 2022-06-30:
You see that I am logged in in the back page but that page asks me to login .. when I do it doesn't accept user name or psw
[image]

luzpaz, 2022-06-30:
on the special:upload page, change the URL to freecad.org to freecadweb.org

luzpaz, 2022-06-30:
@yorikvanhavre we need to do some NGINX-fu to solve this issue 👆

MariwanJ, 2022-07-01:
Thanx @lupaz I close this issue.

closed 2022-07-01 by MariwanJ

luzpaz, 2022-07-04:
Lets make a window of time to switch their domains then?

yorikvanhavre, 2022-07-04:
Good idea.. Mind opening an issue somewhere? (FPA for ex, or homepage...)