MariwanJ/Design456 #81: Uninstalling Design456 WB solved the redundant tabs issue in Prefs>Import/Export.

MariwanJ, 2022-07-12:
as per this page 
https://forum.freecadweb.org/viewtopic.php?f=3&t=70029&p=608101&hilit=design456#p608101
I don't have any linux box. But I found another issue in one file and fixed by 59ee07706c3a66600a90ac83a7373b679fedc2a3

MariwanJ, 2022-07-12:
I keep this just to let FreeCAD people know that the issue is fixed.

MariwanJ, 2022-07-13:
Bug is fixed and announced. close the issue.

closed 2022-07-13 by MariwanJ

luzpaz, 2024-08-22:
Did the change ever make it to the live branch ? We just saw this bug in a ticket :point_up_2:

MariwanJ, 2024-08-23:
yes. you can look at the file in main if you want. Bugs people report, I do change the main directly.

luzpaz, 2024-08-23:
https://github.com/MariwanJ/Design456/blob/be31466f0799651b9330a6377a16309ef08a94fa/Design456Init.py#L68-L73

@MariwanJ is the duplication here necessary ?

MariwanJ, 2024-08-23:
It is a copy of what you find in Draft. Shouldn't affect FC. I do that only in my WB .. 
I need to import DRAFT to Design456.

luzpaz, 2024-08-23:
@MariwanJ it's affecting FC per the bug https://github.com/FreeCAD/FreeCAD/issues/15997 (see the screenshots)

MariwanJ, 2024-08-23:
@luzpaz 
Bug fixed by 320b7e613eb69f03a25e30910c8b1cacffcc9cb4
Thanks for reporting