MariwanJ/Design456 #82: duplicating easyw/Defeaturing_WB

easyw, 2022-10-04:
hi,
I just realized you are including in your wb my Defeaturing_WB, simply duplicating and copying it with same name and functions...
That would lead to conflicts with the original  Defeaturing_WB if installed...
You should at least change name to the wb and to the functions.

MariwanJ, 2022-10-04:
@easyw 
Hi, 
In the other hand, if that WB is included, why should a person install it separately?
There were many bugs that are fixed and code are simplified a bit. 
so, going back to the original .. don't know why?
I try to not modify a lot so the Author doesn't become unhappy that I am hijacking the code. 
Don't know what to say. 
BR

MariwanJ, 2022-10-04:
@easyw 
If you want, make a PR and I will merged. or let me know exactly what I should change.

easyw, 2022-10-04:
> I announced that at my youtube for some times ago.

and then?

> In the other hand, if that WB is included, why should a person install it separately?

your wb started after my wb, moreover instead of forking it, you just 'copied' the whole wb as a subcode of your wb.
Your installation is in conflict with the original wb.

> I try to not modify a lot so the Author (you of course) doesn't become unhappy that I am hijacking the code.

who cares? you would have two options:
1) fork my wb, improve it and make a PR
or
2) derive your wb from mine, but changing the name (i.e.Defeaturing2-wb) and all internal functions names, to avoid conflict with my original wb.

> If you want, make a PR and I will merged.

I will not make a PR... You just decide between option 1) or 2), but change your main wb code.

MariwanJ, 2022-10-05:
@easyw 
Forking rep from other doesn't work for FreeCAD. We discussed that when I tried to fork the pyramid. At the forum, you can find the discussion. It is not possible. And I changed your code to fit to my WB. 

Ok, Give me some time, I will try to fix that today/tomorrow. I need to change many function/class names 
BR

easyw, 2022-10-05:
> Forking rep from other WB and using it as a sub project under another WB doesn't work for FreeCAD.

but you could have forked it and made a PR...
after that you could have included the original WB as a library call inside your WB...
I do it with Manipulator wb inside kicadStepUp wb... no need to duplicate anything.

that way of acting would have have improved both WBs.

MariwanJ, 2022-10-05:
@easyw 
It was natural to combine both WB. I dislike to have so many WB while they have only some little tools. Your tools were good to have in my WB. So, for me it was the way to go. 
But you wish to keep yours. It is up to you. 
I changed almost everything I could. I added o to the start of each function or class. 
Let me know if you have more issues. I close the issue .. Reopen it if you get other issues. 
BR 
Fixed by:
0918a1e58de839f94ac8e16f9845653bb371918e

closed 2022-10-05 by MariwanJ