MariwanJ/Design456 #88: Caused STEP missing colour

meprakarsa, 2024-01-03:
Salaam Mariwan,
As you may have watched issue occurred only when we have "Design456" folder in the "Mod", caused STEP import/open and export missing colours (become only monochrome/gray).

see here: [](https://github.com/realthunder/FreeCAD/issues/742)

[image]

MariwanJ, 2024-01-14:
Alika salam. 
I answered this by my youtube video. 
The issue is not related to my workbench. It is FreeCAD itself. 
look at here 
https://www.youtube.com/watch?v=kbEyfY6CMe4

closed 2024-01-14 by MariwanJ

KeithSloan, 2024-01-15:
Difficult to make out the details of the post in the FC forum in the video, any chance of posting a link?

MariwanJ, 2024-01-15:
I explained in the video, that there are diff import option in FreeCAD which is not explainable. And FreeCAD itself fails from time to time to import the step correctly. 
The menus you use for importing files are written in CPP and my workbench is totally python. I have no access to change anything in the base functionality. i.e. It is a freecad bug and my workbench has nothing to do with the problem. 
You can link this answer at the forum if you want. 

Let them try to import step and chose those different options available. I have no functions regarding step or what ever other file formats. 
Hope I answered your question
Thanks
BR Mariwan