MariwanJ/Design456 #90: Issue with Smart Extrude and Extended Face Functions in Design456 Workbench

Liseun97, 2024-07-01:
Dear MariwanJ,

Hello, I am a student with a strong interest in direct modeling.

I encountered an issue while using the direct modeling feature in the Design456 workbench for FreeCAD. I am reaching out to seek your assistance in resolving this matter.

Firstly, let me describe my working environment.
I am using FreeCAD version 0.21.2.
I have the latest version of Design456, obtained via git clone.
My operating system is Windows 11.

Here are the details of the issue. I experience errors when using features such as Smart Extrude and Extended Face in the Design456 workbench of FreeCAD.

The first screenshot shows the error that occurs when I run Smart Extrude.

[image]
The second screenshot shows the error that occurs when I pull the arrow after selecting Extended Face.
[image]

Could you please provide guidance on how to resolve these errors?

Thank you very much.

Sincerely,
Liseun97

MariwanJ, 2024-07-11:
Hello, If you look at the left widgets, those widget should be put to be as a tab widgets in the left side. yours they are hidden. 
Please look at my videos about how that tabbing mechanism is introduced. 
Regarding the other failure, not sure what you did. Whenever you get such an error. It means that there is something missing or the tool is not properly started from the start. 
Please try that and let me know. 
Sorry for late answer. I am quite busy in the latest 6 month. I might be late in answering you but I will try my best.

MariwanJ, 2024-07-17:
Bug is confirmed. This was working before and it is no longer working.  Should be debugged.

MariwanJ, 2024-07-17:
Bug is fixed. Some import was missing. BTNOK is also moved to be visible. be31466f0799651b9330a6377a16309ef08a94fa

closed 2024-07-17 by MariwanJ

Liseun97, 2024-08-05:
Dear MariwanJ,

I hope this message finds you well. I am writing to express my sincere
gratitude for your previous assistance, which has been incredibly helpful.

I have a few additional inquiries and would greatly appreciate your
guidance on the following matters:

*Hidden Widget Visibility*: Could you kindly provide me with a video link
or any resources that demonstrate how to make a hidden widget visible?
Despite my efforts to find relevant videos and search forums, I have not
been successful in uncovering a clear method. Any direct guidance or
resource would be immensely helpful.

*Extend Face Functionality in Design456*: I am currently studying the
"Extend Face" feature in Design456. My understanding is that when extending
a face, the size is adjusted based on the directional information of the
surrounding edges, similar to the push-pull functionality in Fusion 360.
However, from the attached code, it seems the operation is simply executed
according to the mouse drag size, regardless of the surrounding edge
directions. Could you confirm if I am misunderstanding this process? If so,
could you explain the code responsible for adjusting the face size
according to the surrounding edge directions? Additionally, I would
appreciate it if you could point me to the corresponding code in FreeCAD.

I am extremely grateful for your prompt and detailed responses to my
previous queries.

Thank you once again for your time and assistance.

Best Regards,

Sungeun Lim

*Attachment:*
def MouseDragging_cb(self, userData=None):
    """[Move face by dragging the face. As far as the mouse is pressed.
    This is the callback for all X,Y and Z arrows]
    Args:
        userData ([type], optional): [User Data]. Defaults to None.
    """
    self.StepSize = Design456pref_var.MouseStepSize
    events = userData.events
    if type(events) != int:
        print("event was not int")
        return
    if events != FR_EVENTS.FR_MOUSE_DRAG:
        return #We accept only mouse drag

    if self.discObj.w_userData.Axis_cb is False:
        if self.discObj.w_userData.Disc_cb is True:
            self.RotatingFace_cb(userData)
            return
        else:
            return  # We cannot allow this tool

    self.endVector = App.Vector(self.discObj.w_parent.w_lastEventXYZ.Coin_x,
                                self.discObj.w_parent.w_lastEventXYZ.Coin_y,
                                self.discObj.w_parent.w_lastEventXYZ.Coin_z)
    if self.run_Once is False:
        self.run_Once = True
        # only once
        self.startVector = self.endVector
        self.mouseToArrowDiff =
self.endVector.sub(self.discObj.w_vector[0])/self.StepSize

    deltaChange =
(self.endVector.sub(self.startVector)).dot(self.normalVector)
    self.tweakLength = self.StepSize * (int(deltaChange/self.StepSize))

    if abs(self.oldTweakLength - self.tweakLength) < 1:
        return  # we do nothing
    self.TweakLBL.setText(
        "Length = " + str(round(self.tweakLength, 1)))
    # must be tuple
    self.discObj.label(["Length = " + str(round(self.tweakLength, 1)), ])
    self.discObj.lblRedraw()
    newPos = self.endVector.sub(self.mouseToArrowDiff)
    self.oldFaceVertexes = self.newFaceVertexes
    if self.discObj.w_userData.discObj.axisType == 'X':
        self.newFace.Placement.Base.x = self.StepSize *
(int(newPos.x/self.StepSize))
        self.discObj.w_vector[0].x = self.StepSize *
(int(newPos.x/self.StepSize))
    elif self.discObj.w_userData.discObj.axisType == 'Y':
        self.newFace.Placement.Base.y = self.StepSize *
(int(newPos.y/self.StepSize))
        self.discObj.w_vector[0].y = self.StepSize *
(int(newPos.y/self.StepSize))
    elif self.discObj.w_userData.discObj.axisType == 'Z':
        self.newFace.Placement.Base.z = self.StepSize *
(int(newPos.z/self.StepSize))
        self.discObj.w_vector[0].z = self.StepSize *
(int(newPos.z/self.StepSize))
    else:
        # nothing to do here  #TODO : This shouldn't happen
        return
    self.newFaceVertexes = self.newFace.Shape.Vertexes
    self.COIN_recreateObject()
    self.discObj.redraw()

----------------------------------------------------------
-----------------------------------------------------------------------
임승은 (Seungeun Lim)
M.S. Student, EIF Lab.
School of Mechanical Engineering,
Korea University

Post: 426 Innovation Hall, 145 Anam-ro, Seongbuk-gu, Seoul, 02841, Republic
of Korea (Engineering Information Fusion Lab I)
Mobile : +82-10-8758-8553
E-mail : ***@***.***
--------------------------------------------
-----------------------------------------------------------------------
--------------

MariwanJ, 2024-08-05:
The tool was a start of that project. i.e. modify faces, edges and vertexes with mouse movement. But I failed to complete the effort since FreeCAD engine is way tooo slow. I couldn't continue. Another thing that is problem, is whenever the face is not planner, the modification is quite complex. 
My idea was to stop depending on freecad engine, and do it in mesh. But I am struggling to do so ..

Yes, I am extending the face and the distance is only based on mouse. 

OpenGL platform that can be used to do the job was missing.. I needed to study opengl and learn it. Now I did some progress but unfortunately, i lost my job and still i have no job at the moment. I cannot put my time on the project.
Design456App is to do the direct modeling in MESH not in FreeCAD .. But my plan is/was to make it as a part of freecad .. ie. you convert the object you have to mesh .. work on it .. return back to freecad. 
There is no way to do direct modeling in Freecad while the engine is so slow. Please see the video I have about the topic. 
Now regarding the GUI of Design456 workbench, I will find the link and put it here .. if not I will make a one.

KeithSloan, 2024-08-05:
I wonder if an approach similar to Blender Hard Surface modelling would be a way forward?

Would mean implementing facilities in Blender rather than FreeCAD and have the ability to save as a FreeCAD file. Might be wrong but I got the impression Yorik has developed some code for opening FreeCAD models in Blender.
(Not a great fan of the Blender GUI but lots of people seem to like it)

After the FreeCAD meetings in Brussels and FOSDEM over the weekend a number of developers went on to a first  meeting at Blender on Holland, would love to have gone ( but my expenses did not cover attending the FreeCAD meeting ( Got to find a cheaper place to say than a hotel in the centre of Brussels') maybe they will repeat.

@MariwanJ  hope you find a new job.

MariwanJ, 2024-08-05:
@KeithSloan 
> @MariwanJ hope you find a new job.

Thanks.
Regarding Blender. No, I don't want to go to blender. In my life, I tried several times blender .. but it is not for me. 
Thanks for your wishes.
I need to fix my life before I can return back to Design456

Liseun97, 2024-08-06:
Dear MariwanJ,

I hope this message finds you well.

Thank you very much for your previous email and the materials you sent. I
have carefully reviewed the video you provided, and I will make sure to
display the widget accordingly. Your insights have been invaluable in
helping me identify the areas I need to focus on, particularly in learning
OpenGL and MESH.

Additionally, I would like to ask for your guidance on one more matter. I
understand that you are quite busy, so please feel free to disregard this
request if your schedule does not allow for a response.

I am currently working with the "Extend Face" and "Extend Edge" features of
Design456. From my understanding, these features allow the face or edge to
adjust in size based on the directional information of the surrounding
edges. Could you kindly point me to the part of the code where this is
implemented?

I am eager to study the working principles and code implementation of the
"Extend Face" and "Extend Edge" functions in detail. However, I find myself
struggling to fully comprehend how the resizing mechanism works based on
surrounding edge direction, as illustrated in Design456. Specifically, I am
having difficulty understanding the implementation of the part where the
face or edge adjusts in size when being pulled, as demonstrated in the
examples I have attached (Attachment 1: 0:30–0:35; Attachment 2:
illustrations of resizing when pulling the face or edge). I have also
attached diagrams (Attachment 3) that may clarify what I am referring to.

Your prompt and detailed responses to my previous inquiries have been
incredibly helpful, and I am sincerely grateful for your assistance.

I hope you find a new job soon.

Thank you once again for your support and understanding.

Best Regards,

Sungeun Lim

*Attachment1*: URL: https://www.youtube.com/watch?v=utpurRJ73_8

*Attachment2*: URL: https://www.youtube.com/watch?v=U0Jl5g-OhCQ

*Attachment3*:
 [image: image.png]
----------------------------------------------------------
-----------------------------------------------------------------------
임승은 (Seungeun Lim)
M.S. Student, EIF Lab.
School of Mechanical Engineering,
Korea University

Post: 426 Innovation Hall, 145 Anam-ro, Seongbuk-gu, Seoul, 02841, Republic
of Korea (Engineering Information Fusion Lab I)
Mobile : +82-10-8758-8553
E-mail : ***@***.***
--------------------------------------------
-----------------------------------------------------------------------
--------------

MariwanJ, 2024-08-06:
Hi @Liseun97, 
I will try to explain the whole concept. 
1-For planner surfaces : 
You need to find all vertexes that will be modified. Change their location based on the mouse movements. Those arrows and discs are made with Coin3D. Coin3D is complex and needs it's own study. After moving each step the vertexes, you have to rebuild the shape as whole. Now, this does not work with FreeCAD as it is too slow. What you see is a fake shape I make in Coin3D and later I reconstruct the original shape with the new vertexes. 
2-Curved surfaces: This is very complicated. You need to know the shape itself .. how would you change that? if it is a part of a cylinder, you have an option or two, if it is a curved surface .. that means you need a way to do the modification. This is very hard and I really didn't yet manage to do it. For such curved surface, you need to have multiple options. 

simply, based on vertex positions, you have to recreate a modified version of your surface or shape. There are other way to do the whole concept and that should be implemented to the root of the shapes. i.e. you never join the original shapes you have (box, cylinder ..etc) and when you modify the surface .. you actually modify the original shape by resizing or chaining radius ..etc. 
That concept needs to change FreeCAD document totally which is not possible. 

Direct modeling is hard .. not easy task. Doing it in Mesh might simplify the work alot .. that is why most of the software actually uses some kind of MESH IMHO. 

hope I explained for you little bit. I am really sorry, I am not able to put much more time to answer in depth. BR /Mariwna